ropg/heltec_esp32_lora_v3 #1: WirelessStick behavior.

aintgotnoname, 2024-03-07:
Thank you for this.  I stumbled across the repository within an hour of your initial commit.  It has been quite helpful.  I'm using the WirelessStick (V3) with minor "issue".  More nit-picky nonsense.

setting the display geometry to GEOMETRY_64_32 with a "WirelessStick" define allows the micro display to function.

The led does not turn off with 0% specified (>0 vs >=0).

Some examples redundantly 'flip the display vertically' duplicating (neutering) the operation in heltec.h

would be nice if heltec.h worked as a pure include 'extern'ing functions/objects rather than declaring/instantiating them.  Using multiple files within platformio projects isn't so simple.

again, thank you. wonderful work beyond these very minor nits.  not being github literate, I don't know how to push stuff up.

ropg, 2024-03-07:
> Thank you for this. I stumbled across the repository within an hour of your initial commit. It has been quite helpful. I'm using the WirelessStick (V3) with minor "issue". More nit-picky nonsense. [...]

I don't have a WirelessStick, but I'll make that #define, sure. Anything else that needs to change for the WirelessStick to work with all the functions. Battery voltage measuring, etc, everything all the same? (Naturally my examples will need work then too. Should get me a Stick, I guess.

> The led does not turn off with 0% specified (>0 vs >=0).

```cpp
    // LED
    for (int n = 0; n <= 100; n++) { heltec_led(n); delay(5); }
    for (int n = 100; n >= 0; n--) { heltec_led(n); delay(5); }
    display.println("LED works");
```

This bit in the demo program in the README does turn off the LED. The `pinMode(LED_PIN, INPUT)` (if that's what you mean) is so that the pin is not driven when the LED is off. Which might save a teeny-tiny bit of power.
 
> Some examples redundantly 'flip the display vertically' duplicating (neutering) the operation in heltec.h

Taken those out, thanks.

> would be nice if heltec.h worked as a pure include 'extern'ing functions/objects rather than declaring/instantiating them. Using multiple files within platformio projects isn't so simple.

That sort of conflicts with my desire to make things 'just work', but I get it. But if you upgrade to 0.3.2 and `#define HELTEC_NO_INSTANCES` before you include the library, you should get what you want.

> again, thank you. wonderful work beyond these very minor nits. not being github literate, I don't know how to push stuff up.

Thanks, happy you like it.

closed 2024-03-07 by ropg

ropg, 2024-03-07:
Still wondering about the LED not turning off. Was that observed in practice, or just from reading the code? If in practice, what was the code?

aintgotnoname, 2024-03-07:
Observed.  Clearly, the original code left the led quite visible at 0. (I didn't try 1%)  
I then jumped in and changed to allow 0 to drive the led off, and it did just that.

I figured, given the input was an integer, the original intention is maintained with 0-100 driving the led; and -1 (or less) floating the pin.  Hence, my humble suggestion.

void heltec_led(int percent) {
  if (percent >= 0) {
    ledcSetup(LED_CHAN, LED_FREQ, LED_RES);
    ledcAttachPin(LED_PIN, LED_CHAN);
    ledcWrite(LED_CHAN, percent * 255 / 100);
  } else {
    pinMode(LED_PIN, INPUT);
  }
}

aintgotnoname, 2024-03-08:
I don't know.  Mea culpa?  The absolute minimum demo sketch works as anticipated (as originally written).

I was mucking around with the rx tx demo when originally experiencing led issue.