ropg/heltec_esp32_lora_v3 #4: [TYPO] Wrong link to RTC EEPROM in README.md

Velocet, 2024-03-11:
Don't wanna fork everything for just a typo ;)

[https://github.com/ropg/heltec_esp32_lora_v3/blob/main/README.md?plain=1#L57](https://github.com/ropg/heltec_esp32_lora_v3/blob/main/README.md?plain=1#L57)

I only have the v2 version but it's good to see someone else also hates HelTecs documentation...

ropg, 2024-03-11:
Eh.. Sorry... I must be missing what the typo is...

closed 2024-03-13 by ropg

C3PHLOP0D, 2024-03-13:
The typo - the URL linked is bad. It is missing a / after the .com.

https://github.com.ropg/ESP32_RTC_EEPROM

ropg, 2024-03-14:
Ah, I get what happened. Your message included a link to a specific line, which only said `#include <heltec.h>`. So I couldn't see what was wrong with that. When you sent it, that line probably contained the offending typo. Silly GitHub.

Fixed. Thanks!