ropg/heltec_esp32_lora_v3 #7: HotButton.h missing from distribution

tao480, 2024-03-21:
The file HotButton.h missing seems to be missing from src directory.

tao480, 2024-03-21:
Yes, certainly odd.  Interestingly, I also had an issue with another dependency that wasn't automatically installed by the Arduino IDE.  

The problem was caused by a different expressif Additional Boards Manager URL to the Github version you had recommended and referenced, namey https://espressif.github.io/arduino-esp32/package_esp32_index.json.  

I've now lost the previous link to package_esp32_index.json, however it was the one hosted on expressif's own domain. 

This is the dependency that was failing:
 {
              "packager": "esp32",
              "name": "openocd-esp32",
              "version": "v0.12.0-esp32-20230419"
            },

The expressif hosted version of package_esp32_index.json didn't include the 19Apr2023 version of the openocd-esp32 tool set.  The problem was fixed by changing the expressif boards manager URL to the Github version: https://espressif.github.io/arduino-esp32/package_esp32_index.json, however I also had to switch to Arduino IDE version 2.x to force it to execute the download.

ropg, 2024-03-22:
The Arduino IDE can be strange: I've had to restart it two times sometimes before it wanted to load new board defs. The way boards are defined is arcane and does not reflect how the world works in 2024: this notion that a manufacturer like Espressif will somehow disseminate up-to-date definitions for all hardware with an ESP32 in it. Hmmm

closed 2024-04-12 by ropg