ropg/heltec_esp32_lora_v3 #8: Board installation doesn't seem to work on Arduino IDE V1.8.x

tao480, 2024-03-21:
The "Heltec (unofficial)" board doesn't appear in the Boards Manager when using Arduino IDE V1.8.x.

After identifying this issue I managed to install the board successfully using Arduino IDE 2.x.x and then reverted to Arduino IDE V1.8.x, however, the "Heltec (unofficial)" board still didn't appear.

ropg, 2024-03-21:
Everything here is tested on Arduino IDE 2.3.2, and the board appears here. Out of curiosity: why do you insist on 1.8 ?

tao480, 2024-03-21:
Rop, I already had a lot of ESP32 code tested and working on Arduino IDE 1,8.x, so I didn't want to risk destablising any of that code.

Also, I've been finding the 2.x user interface rather clumsy and time consuming.  However, I guess I'll have to make the switch at some point; either that or move to Platform.io

tao480, 2024-03-21:
Rop, I forgot to add my thanks for the great job you've done in cleaning up the very unsatisfactory Heltec libraries for display, radio and the press button.

Also, another oddity I encountered was the compiler reporting a duplicate instance of the heltec() entry point.  The problem was fixed by removing all the old references I had to the Heltec LoRaWAN library and by also hard coding the path to your library by changing the #include directive, to this:
#include "D:\Arduino\libraries\heltec_esp32_lora_v3-main\src\heltec.h"

capitancafe, 2024-03-21:
Hi Rop, thank´s for your Job. I cannot install in Ide 2.3.2 because i´ve this error:
Failed to install platform: 'Heltec (unofficial):1.1.3'. 9 FAILED_PRECONDITION: Platform 'Heltec (unofficial):esp32@1.1.3' not found: getting tool dependencies for platform Heltec (unofficial):esp32@1.1.3: tool version v0.12.0-esp32-20230419 not found

capitancafe, 2024-03-21:
Solved! I have not intalled ESP32 libraries for Espressif

ropg, 2024-03-22:
@capitancafe Still shouldn't give that error, good to know there's some element of the toolchain that doesn't get copied. Tool must be in my path somewhere.

@tao480 Yeah, I could have avoided the problem with the other heltec library if I had just named mine something different. I just don't have the official heltec stuff installed.

tao480, 2024-03-22:
Rop, I think the following official Arduino article describes why the Boards Manager isn't working in Arduino IDE V1.x
https://arduino.github.io/arduino-cli/0.33/package_index_json-specification/

In particular, the following sections seems to be relevant to the issue:
_The file must be named as follows:

package_YOURNAME_PACKAGENAME_index.json

The prefix package_ and the postfix _index.json are mandatory (otherwise the index file is not recognised by the Arduino development software) while the choice of YOURNAME_PACKAGENAME is left to the packager._

tao480, 2024-03-22:
OK Rop, so a bit more confirmation re the Additional Boards Manager issues, especially for Arduino IDE 1.6 (onwards) users.

As per my previous message, the Additional Boards Manager URL must be formatted as "package_<your_name_here>_index.json".  To test this I downloaded your boards.json file, renamed it (as described) and then posted it to my web server.  I then added my the link to my web server's version of the the package (above the link you provided for your URL), saved it, visited the Boards Manager and your package downloaded successfully.

Then the next issue I found in my test Arduino ISE 1.8.x installation was that your package appears under the Boards Manager menu as "ESP32 Arduino".  A little annoying, but it works. Your three supported boards appear correctly in the sub-menu.  

On further investigation, line 1 of your platforms.txt, which currently reads "name=ESP32 Arduino", seems to be the cause of this minor annoyance.

tao480, 2024-03-22:
Confirming that line 1 of C:\Users\<user_id>\AppData\Local\Arduino15\packages\Heltec (unofficial)\hardware\esp32\1.1.3\platform.txt is the source of the text that appears on the Boards Manager drop-down menu.

I've changes line 1 of that file to "name=Heltec (unofficial)" and that change is immediately reflected in the Boards Manager drop-down.

ropg, 2024-03-22:
I'll have a look at that later this weekend. Want to change it so people can menu-select soem RadioLib protocol debug output and a few other things anyway.

cjn566, 2024-03-23:
> Solved! I have not intalled ESP32 libraries for Espressif

Can you tell me what libraries those are? I'm running into the same issue: 
`Error: 9 FAILED_PRECONDITION: Platform 'Heltec (unofficial):esp32@1.1.3' not found: getting tool dependencies for platform Heltec (unofficial):esp32@1.1.3: tool version v0.12.0-esp32-20230419`

tao480, 2024-03-23:
@cjn566 Be sure your Additional Boards Manager (File->Preferences) is pointing to Expressif's Github site (https://espressif.github.io/arduino-esp32/package_esp32_index.json) and NOT the version of the same thing they also host on their own domain, which doesn't include the 19Apr2023 of the tools.

ropg, 2024-03-23:
The thing is copied the espressif files and (or so I thought) only deleted the toolchains for the ESP32 chips not in use here: i.e. everything but the S3. I can reproduce the problem now, and I'll hopefully fix it without the need for anything else to be installed.

davidsaintruby, 2024-03-24:
I just want to say thanks for working on this.  I hate the Heltec boards, everyone one of them is library torture, and you are one of the saints trying to fix it.

cjn566, 2024-03-24:
Alright I've made progress. After adding the espressif board definitions, I was able to get yours installed.

Now I'm trying to upload the minimal demo sketch and it's failing:
`esptool.py v4.5.1
Serial port COM1
Connecting......................................

A fatal error occurred: Failed to connect to ESP32-S3: No serial data received.`

Sorry to be such a noob, is there a trick to it?

tao480, 2024-03-24:
@cjn566 Sounds like you're making slow and steady progress.
There are some ideas here that may be helpful.  https://forum.arduino.cc/t/solved-esp32-s3-cannot-upload-the-sketch/1058691/8

capitancafe, 2024-03-25:
@ropg ¿Do you have a example for run as Lorawan gateway? the code of https://github.com/things4u/ESP-1ch-Gateway/blob/master/src/loraModem.h not work, i think because the lora chip it´s not the same, and some pins arent connected.

ropg, 2024-04-12:
I'm testing a new method of adding the board definitions. Pease check out #14 if you want to help test it.

closed 2024-04-12 by ropg