ropg/heltec_esp32_lora_v3 #12: Arduino 2.3.2 error installing board

mludlum, 2024-04-11:
After following your instructions, I did not see the Unofficial option under Tools->Board...

I went to board manager, searched for Heltec, and saw Heltec (unofficial) by Rop.  When I clicked Install for version 1.1.5, I get this popup:

Failed to install platform: 'Heltec (unofficial):1.1.5'. 9 FAILED_PRECONDITION: Platform 'Heltec (unofficial):esp32@1.1.5' not found: getting tool dependencies for platform Heltec (unofficial):esp32@1.1.5: tool version v0.12.0-esp32-20230419 not found

mludlum, 2024-04-11:
Windows 11 Pro

mludlum, 2024-04-11:
Can you tell me if you have an esp32 boards installed and which versions?  

I'm looking at this thread:  https://forum.arduino.cc/t/tool-release-not-found-arduino-bossac-1-3a-arduino/1076336/3
which gives an explanation for a similar problem.  In this case, the package index 
https://github.com/ropg/heltec_boards/blob/main/package_heltec_unofficial_index.json

is expecting the packager esp32 to have version v0.12.0-esp32-20230419

But the latest version here is 2024...
https://github.com/espressif/openocd-esp32/releases

Not sure if I am getting closer to the solving this or confused.   I tried installing Arduino and Espressif esp32 boards but that didn't work.

ropg, 2024-04-12:
I'm testing a new method of adding the board definitions. Pease check out #14 if you want to help test it.

closed 2024-04-12 by ropg