ropg/heltec_esp32_lora_v3 #15: Unable to use with Arduino IDE 2.3.2

gradew, 2024-04-17:
I've only been able to use this board library with the legacy version of the Arduino IDE (1.8.19), albeit with a warning.
When trying to compile with the latest version of the IDE (2.3.2), I'm getting the following error message:

Invalid FQBN: tool release not found: esp32:openocd-esp32@v0.12.0-esp32-20230419
Compilation error: Invalid FQBN: tool release not found: esp32:openocd-esp32@v0.12.0-esp32-20230419

It seems esp32-20230419 is missing from the boards.json definition.

ropg, 2024-04-17:
This is known issue that I'm fixing as soon as I have some time this week. [Here](https://github.com/ropg/heltec_esp32_lora_v3/discussions/14)'s the discussion about the new method.

closed 2024-04-17 by ropg

gradew, 2024-04-17:
Hi,

Thanks for your response, and my apologies for creating this issue: I didn't see the discussion about the workaround (which works great !)

Cheers