ropg/heltec_esp32_lora_v3 #17: Basic example failure

xznhj8129, 2024-04-27:
In file included from /tmp/arduino_modified_sketch_288695/LoRa_rx_tx.ino:16:
/home/anon/Arduino/libraries/Heltec_ESP32_LoRa_v3/src/heltec_unofficial.h:14:10: fatal error: driver/temp_sensor.h: No such file or directory
 #include "driver/temp_sensor.h"
          ^~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
exit status 1
Error compiling for board Heltec Wireless Stick Lite.

ropg, 2024-04-27:
Are you using the Arduino IDE? If so, what version on what OS? How did you install ESP32 support?

xznhj8129, 2024-04-27:
I installed it with espressif board manager URL and also the heltec install method, but I found my persistant issue, i just had to use the latest IDE. I had to use 1.8 for another project. Consider this solved.

closed 2024-04-27 by xznhj8129