ropg/heltec_esp32_lora_v3 #18: heltec_deep_sleep wrapper(): absence of cast of "seconds" to (u_int64_t) leads to overflow

gradew, 2024-04-27:
When trying to get my ESP32 to sleep for an entire day (3600 seconds times 24, 86400 seconds), I noticed it would wake up roughly every 8 minutes.
After looking at the code for the heltec_deep_sleep() wrapper, I noticed the "seconds" argument is an int (signed 32-bit integers on the ESP32) that gets multiplied by 1000000 before being passed on to esp_sleep_enable_timer_wakeup() without being cast to an int64_t.
The result of that multiplication should be 86400000000us, but it ends up being 500654080us because of the 32-bit overflow, which, lo and behold, is 8.333 minutes :)

I can do a PR if you're busy/interested.

Very cool library, I love it :)

ropg, 2024-04-28:
Glad you like the library. And well spotted. Whoops. Fixed it, and released an update immediately because that's an annoying bug.

closed 2024-04-28 by ropg

gradew, 2024-04-28:
That was quick :) Thanks a lot, and keep up the good work \m/