ropg/heltec_esp32_lora_v3 #19: display.clear() doesn't seem to work

JakeBiggs, 2024-05-05:
When calling
display.clear();
the screen does not clear previously written text.
Upon further testing, it seems the display.clear works in the simple demo example, but the UIexample is also broken

JakeBiggs, 2024-05-05:
Use display.cls(); if using text and should work fine

closed 2024-05-05 by JakeBiggs

ropg, 2024-05-05:
Yup, `display.clear()` clears the graphics buffer, which does not get updated until you call `display.display()`. `display.cls()` clears the display immediately and also clears the text buffer. Admittedly confusing, but it's in the [documentation](https://github.com/ThingPulse/esp8266-oled-ssd1306#api).

JakeBiggs, 2024-05-06:
Yes I had a function:

void clearDisplay(){
    display.clear();
    display.display();
}

And it still didn't work so i'll just stick to cls() for now. I'm not sure if it's an address issue or whatever, but thanks for the library man! super helpful :)

ropg, 2024-05-06:
> Yes I had a function:
>
> [...]

That's odd, will try to look into that at some point when doing something with the display library anyway. Glad you like the library.