ropg/heltec_esp32_lora_v3 #20: DevNonce isnt being saved

recklessop, 2024-05-08:
Recently I updated all the libraries to the latest builds. Now on a second packet to Chirpstack the Heltec  clearly forgets the data that was supposed to be moved from rtc ram to EEPROM. When I used the library version from several weeks ago, this worked fine. So I'm just looking for help to see if my program might need a tweak, or if something in the library changed.

Here is the output from the code. 

First Run: 
rst:0x1 (POWERON),boot:0x28 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x4bc
load:0x403c9700,len:0xbd8
load:0x403cc700,len:0x2a0c
entry 0x403c98d0

Soil 0.51v
Battery 4.06v 94.00%
RTC RAM -> flash.
Next TX in 300 s

Second run:
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x28 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x4bc
load:0x403c9700,len:0xbd8
load:0x403cc700,len:0x2a0c
entry 0x403c98d0
Radio init
[RadioLib] radio.begin() returned 0 (ERR_NONE)
Joining (forced)
[RadioLib] node.beginOTAA(joinEUI, devEUI, nwkKey, appKey, RADIOLIB_LORAWAN_DATA_RATE_SF_12, true) returned -1116 (See https://jgromes.github.io/RadioLib/group__status__codes.html)
RTC RAM -> flash.
Next TX in 300 s

here is the program i am running:

#include <CayenneLPP.h>
#include <heltec_unofficial.h>
#include <LoRaWAN_ESP32.h>
#include <ESP32_RTC_EEPROM.h>

/*
  Deep Sleep LoRaWAN TTN example

  This example assumes the ESP32 stays powered, but puts it in a very low-power
  "deep sleep" mode between sends to the network.

  This example joins a LoRaWAN network and will send uplink packets. Before you
  start, you will have to register your device at
  https://www.thethingsnetwork.org/ After your device is registered, you can run
  this example. The device will join the network and start uploading data.

  LoRaWAN v1.1 requires storing parameters persistently. RadioLib does this by
  using the Arduino EEPROM storage. The ESP32 does not have EEPROM memory, so
  the Arduino code from Espressif has code that emulates EEPROM in its "NVS"
  flash key-value storage. This unfortunately means that every change of a
  single but re- writes the entire emulated "EEPROM" to flash again. 

  That would be great for the occasional configuration change, but we need to
  keep state after every single interaction with the LoRaWAN network. Flash
  memory can only be erased and written to a limited number of times. This is
  why this library contains a fork of RadioLib that uses ESP32_RTC_EEPROM, which
  writes to the memory of the ESP32's onboard Real-Time Clock which is kept
  powered during the chip's "deep sleep", which uses almost no power.

  This example then backs up that RTC RAM memory to flash
    a) every time it is woken up with a wiped RTC RAM,
    b) whenever it receives a downlink packet, or 
    c) every so many times as set by BACKUP_EVERY.
[... 234 lines cut ...]
  // And then pick it or our MINIMUM_DELAY, whichever is greater
  uint32_t delayMs = max(interval, (uint32_t)MINIMUM_DELAY * 1000);
  
  both.printf("Next TX in %i s\n", delayMs/1000);

  // Give the user a chance to see the message
  //both.println("Deep sleep in 5 s");
  delay(5000);

  // And off to bed we go
  heltec_deep_sleep((delayMs / 1000) - 5);

}

float readBatteryVoltage() {
  int raw = analogRead(A0);
  float voltage = raw / 4095.0 * 3.3;  // Convert ADC value to voltage
  return voltage;
}

float readSoilMoisture() {
  int raw = analogRead(A6);
  float voltage = raw / 4095.0 * 3.3;
  return voltage;
}

ropg, 2024-05-08:
Forget anything that tries to save to EEPROM, and try (basing off) the example that comes with the [LoRaWAN_ESP32 library](https://github.com/ropg/LoRaWAN_ESP32). Or, if you really want to hardcode credentials, the modified example [here](https://github.com/ropg/LoRaWAN_ESP32/issues/1#issuecomment-2080394146).

ropg, 2024-05-10:
Did you work it out, or do you need further help?

ropg, 2024-05-13:
Closing issue, feel free te reopen if you need further help.

closed 2024-05-13 by ropg

xistoso, 2024-05-18:
Sorry to comment in a closed issue. But i'm facing the same issue. 
Looks like the library is repeating the "devnounce" and chirpstack has the hability to remember a few devnouces. How are you generating those random values?
I can only communicate like 2 - 5 times before facing this issue. And needing to flush the nouces in chirpstack side.

ropg, 2024-05-19:
Please post the code used and I'll have a look.