ropg/heltec_esp32_lora_v3 #21: ESP32 Lora v3 spectrum scanner

Aerostar666, 2024-05-12:
Hi, Sorry but I'm unable to find the file SX126x_patch_scan.h anywhere.

Is there a special place I should look into ?

Thanks :-)

Velocet, 2024-05-12:
Can you please first describe your problem in a better way? When/Where it happens or even some log? This would help you to get a better answer :) 

The file in question is part of RadioLib and could be found here: [SX126x_patch_scan.h](https://github.com/jgromes/RadioLib/blob/master/src/modules/SX126x/patches/SX126x_patch_scan.h)

GitHub also has a search feature:
[path:**/SX126x_patch_scan.h](https://github.com/search?q=path%3A**%2FSX126x_patch_scan.h&type=code)

Aerostar666, 2024-05-12:
Thank you for pointing the way :-) The file is simply not included in the various libraries in Arduino IDE.
Even in the Radiolib library, this file seems to be missing.
I'll try to add it manually.
Thank you :-)

ropg, 2024-05-13:
Ah, found the problem. My path in the code still had "Radiolib/" prefixed from when I included a fork of RadioLib. With that removed it finds the file in the proper RadioLib directory. Simply upgrade to 0.8.2 to get this fix.

closed 2024-05-13 by ropg