ropg/heltec_esp32_lora_v3 #23: Not compatible with esp32 3.0.0-rc1 library?

ademars, 2024-05-16:
Is the library supposed to work with the current esp32 3.0.0-rc1 libraries? I’m having build issues (ie can’t find pins_arduino.h).

ropg, 2024-05-17:
"tsens: CONFLICT! driver_ng is not allowed to be used with the legacy driver"

Ah, that would be the temperature sensor code. I'll fix that with an #ifdef. But until then as a workaround if you do not need ESP32 chip temperature, just comment out the entire function `float heltec_temperature()`

Please try again and let me know if that's all that needs fixing.

ademars, 2024-05-17:
That fixed the infinite boot loop. Thanks! The LoRa_rx_tx sketch now builds, runs, and interoperates with the heltec provided “WiFi_LoRa_32_V3_FactoryTest“ sketch. However, it appears that the “heltec_led” API isn’t working. The led never comes on. If I simply use digitalWrite(LED, 1/0), I see the led blink on each transmit. Alan

Velocet, 2024-05-17:
Please use [LED Control (LEDC)](https://espressif-docs.readthedocs-hosted.com/projects/arduino-esp32/en/latest/api/ledc.html)

ropg, 2024-05-20:
In new version 0.9.0 I just released, both temperature sensing and LED driver were fixed to work with both v2.0.16 and v3.0.0 of ESP32 Arduino.

closed 2024-05-20 by ropg