ropg/heltec_esp32_lora_v3 #24: PlatformIO VSC config Ubuntu

jo-ei, 2024-05-26:
Hi,
i like to use it on my Ubuntu with Visual Studio Code with platform.io.

here is my platformio.ini config
```
[platformio]
default_envs = my-config
description = my Test Project

[env:my-config]
platform = espressif32
board = heltec_wifi_lora_32_V3
framework = arduino
monitor_speed = 115200

lib_deps = 
	ropg/Heltec_ESP32_LoRa_v3@^0.9.0

build_flags = 
	-DWireless_Stick_V3=1
	-DREGION_EU868=1
	-DHELTEC_WIRELESS_STICK_LITE=1
```

i got the following error:
```
In file included from src/main.cpp:5:
.pio/libdeps/my-config/Heltec_ESP32_LoRa_v3/src/heltec_unofficial.h:33:19: error: 'const uint8_t GPIO_NUM_8' redeclared as different kind of symbol
 #define SS        GPIO_NUM_8
                   ^~~~~~~~~~
/home/user/.platformio/packages/framework-arduinoespressif32/variants/heltec_wifi_lora_32_V3/pins_arduino.h:23:22: note: in expansion of macro 'SS'
 static const uint8_t SS    = 8;
                      ^~
In file included from /home/user/.platformio/packages/framework-arduinoespressif32/tools/sdk/esp32s3/include/esp_hw_support/include/esp_sleep.h:13,
                 from /home/user/.platformio/packages/framework-arduinoespressif32/cores/esp32/esp32-hal.h:33,
                 from /home/user/.platformio/packages/framework-arduinoespressif32/cores/esp32/Arduino.h:36,
                 from .pio/libdeps/my-config/RadioLib/src/BuildOpt.h:121,
                 from .pio/libdeps/my-config/RadioLib/src/TypeDef.h:6,
                 from .pio/libdeps/my-config/RadioLib/src/RadioLib.h:38,
                 from .pio/libdeps/my-config/Heltec_ESP32_LoRa_v3/src/heltec_unofficial.h:53,
                 from src/main.cpp:5:
/home/user/.platformio/packages/framework-arduinoespressif32/tools/sdk/esp32s3/include/hal/include/hal/gpio_types.h:240:5: note: previous declaration 'gpio_num_t GPIO_NUM_8'
     GPIO_NUM_8 = 8,     /*!< GPIO8, input and output */
     ^~~~~~~~~~
```

Same by the other: MOSI,  MISO, SCK, RST_OLED, BUSY_LoRa, .......

Can somebody help me to get this fixed?

Thanks

Velocet, 2024-05-28:
build_flags = 
	-DWireless_Stick_V3=1
	-DREGION_EU868=1
	-DHELTEC_WIRELESS_STICK_LITE=1

You defined the used board three times. Try this instead:

build_flags = 
	-DREGION_EU868=1

jo-ei, 2024-05-28:
Thanks for the tip, unfortunately the error remains 100% the same.

According to the documentation "5. Stick or Stick Lite?" I have to set the value HELTEC_WIRELESS_STICK_LITE.

I'm currently just not sure if -DREGION_EU868=1 works for the frequency selection

ropg, 2024-05-28:
To my knowledge, that region definition only does something if you use the stack that Heltec included with its library. As we use RadioLib, it does nothing.

ropg, 2024-05-30:
As for having everything in the .h file:

I gave that some thought actually, but simply could not come up with a scenario where you'd need/want to include this library twice, ever. There are advantages to doing it this way, the main one being that I can make the code behave differently based on `#define` statements before the `#include`. That doesn't work if the library is a separate compilation unit.

How would you then turn off compilation of the display driver, or tell `loop()` to do button power-off? Modify a config.h in library dir? (harder to explain, possibly overwritten on upgrades)

ropg, 2024-05-30:
As for the definitions. I put them in there so that things still worked when playing with generic ESP32 board definitions while experimenting with my own board definitions, a direction I have since abandoned because it was a much larger can of worms than I thought. It's on my list to make things work with platformio, just have been busy with other things.

jo-ei, 2024-07-07:
my working config at the moment

```
[platformio]
default_envs = heltec
description = my Test Project

[env:heltec]
; https://github.com/platformio/platform-espressif32/releases/tag/v6.1.0
platform = espressif32@>=6.1.0
board = heltec_wifi_lora_32_V3
framework = arduino
monitor_speed = 115200
monitor_filters =
	default
	esp32_exception_decoder

lib_deps = 
	ropg/Heltec_ESP32_LoRa_v3@^0.9.1
	ropg/LoRaWAN_ESP32@^1.1.0

build_flags = 
	-DNO_DISPLAY
```

closed 2024-07-07 by jo-ei