ropg/heltec_esp32_lora_v3 #27: 'class LoRaWANNode' has no member named 'isJoined'

sledzik1984, 2024-05-30:
`// Pause between sends in seconds, so this is every 15 minutes. (Delay will be
// longer if regulatory or TTN Fair Use Policy requires it.)
#define MINIMUM_DELAY 900 
#define HELTEC_WIRELESS_STICK

#include <heltec_unofficial.h>
#include <LoRaWAN_ESP32.h>

LoRaWANNode* node;

RTC_DATA_ATTR uint8_t count = 0;

const char* band       = "EU868";
const uint8_t subband  = 0;
const uint64_t joinEUI = 0x0000000000000000;
const uint64_t devEUI  = 0xxxxx;

const uint8_t appKey[] = { 0xC1,  };
const uint8_t nwkKey[] = { 0x86, };

void setup() {
  heltec_setup();

  // Obtain directly after deep sleep
  // May or may not reflect room temperature, sort of. 
  float temp = heltec_temperature();
  Serial.printf("Temperature: %.2f °C\n", temp);

  // Put the provisioning data in flash. Can do this every time: flash values
  // are only ever written if they differ from what's already there.
  persist.provision(band, subband, joinEUI, devEUI, appKey, nwkKey);

  // initialize radio
  Serial.println("Radio init");
  int16_t state = radio.begin();
  if (state != RADIOLIB_ERR_NONE) {
    Serial.println("Radio did not initialize. We'll try again later.");
    goToSleep();  // Does not return, program starts over after sleep
  }

  node = persist.manage(&radio);

  if (!node->isJoined()) {
    Serial.println("Could not join network. We'll try again later.");
    goToSleep();  // Does not return, program starts over after sleep
  }

  // If we're still here, it means we joined, and we can send something

  // Manages uplink intervals to the TTN Fair Use Policy
  node->setDutyCycle(true, 1250);

  uint8_t uplinkData[2];
  uplinkData[0] = count++;
  uplinkData[1] = temp + 100;

  uint8_t downlinkData[256];
  size_t lenDown = sizeof(downlinkData);

  state = node->sendReceive(uplinkData, sizeof(uplinkData), 1, downlinkData, &lenDown);

  if(state == RADIOLIB_ERR_NONE || state == RADIOLIB_ERR_RX_TIMEOUT) {
    Serial.println("Message sent");
  } else {
    Serial.printf("sendReceive returned error %d, we'll try again later.\n", state);
  }

  goToSleep();    // Does not return, program starts over after sleep

}

void loop() {
  // This is never called. There is no repetition: we always go back to
  // deep sleep one way or the other at the end of setup()
}

void goToSleep() {
  Serial.println("Going to deep sleep now");
  // allows recall of the session after deepsleep
  persist.saveSession(node);
  // Calculate minimum duty cycle delay (per FUP & law!)
  uint32_t interval = node->timeUntilUplink();
  // And then pick it or our MINIMUM_DELAY, whichever is greater
  uint32_t delayMs = max(interval, (uint32_t)MINIMUM_DELAY * 1000);
  Serial.printf("Next TX in %i s\n", delayMs/1000);
  delay(100);  // So message prints
  // And off to bed we go
  heltec_deep_sleep(delayMs / 1000);
}`

Tryint to compile that code ends with 

`/Users/piotr/Documents/Arduino/Arduino-sketches/LoRaWan-Skrzynka/LoRaWan-Skrzynka/LoRaWan-Skrzynka.ino: In function 'void setup()':
/Users/piotr/Documents/Arduino/Arduino-sketches/LoRaWan-Skrzynka/LoRaWan-Skrzynka/LoRaWan-Skrzynka.ino:43:14: error: 'class LoRaWANNode' has no member named 'isJoined'
   if (!node->isJoined()) {
              ^~~~~~~~

exit status 1

Compilation error: 'class LoRaWANNode' has no member named 'isJoined'`

Can anyone assist me with that issue?

jo-ei, 2024-05-31:
hi,
i got the same error.

looks like it is no longer a keyword in [RadioLib]/keywords.txt
you can find the "isJoined" in a older version from keywords.txt.

sledzik1984, 2024-05-31:
So we have to use old Radiolib version?

At the moment with every library updated example LoraWAN_TTN will not compile.

There is nothing in the docs that we need to use exact version of some libraries

jo-ei, 2024-05-31:
> So we have to use old Radiolib version?

No i think it is better to wait for an update here from @ropg

icepick3000, 2024-05-31:
Same problem here.. the trouble started when i updated my radiolib.. I am going to try to downgrade it.

ropg, 2024-05-31:
The new RadioLib version has just been released, and they changed the LoRaWAN API. I will update this library very soon, probably within a few days. Until then, downgrade to RadioLib 6.5.0.

closed 2024-05-31 by ropg

ropg, 2024-06-04:
Upgrade to 0.9.1 of this library, version 1.1.0 of LoRaWAN_ESP32 and version 6.6.0 of RadioLib and the TTN example should work again.

icepick3000, 2024-06-14:
Yes, works again, thanks!