ropg/heltec_esp32_lora_v3 #28: multiple definition

jo-ei, 2024-06-08:
Hi,
i go the following error in VSC with platformio:

`multiple definition of _Z10heltec_ledi; `
i got these for: heltec_led, heltec_ve, heltec_vbat, heltec_deep_sleep, heltec_setup

Somehow I'm not sure where that comes from.
can anyone give me a tip on where I am include something wrong?

thx

aintgotnoname, 2024-06-10:
both "lorawan.cpp" and "main.cpp" directly or indirectly include "heltec_unofficial.h" which contains the definition of the code block for heltec_led() and its siblings.  it seems the linker treats each include reference as a duplicate definition of the same block; thus generating an error(s).

imho, you will need to determine a means of declaring these references to a single inclusion of heltec_unofficial.h within your project.  this may be performed by littering the code with extern declarations of the functions, or creation of an additional include file with the function prototypes.  other workarounds may exist, but are not known to me.

Velocet, 2024-06-11:
> [env:my-config]
> board = heltec_wifi_lora_32_V3
>
> build_flags = 
> 	-D Wireless_Stick_V3=1
> 	-D HELTEC_WIRELESS_STICK_LITE=1

Why are there three different board definitions? Choose one and stick to it or better: don't use build flags you don't know.

If you really have a LoRa32v3 then delete those two build flags.

closed 2024-07-21 by jo-ei