ropg/heltec_esp32_lora_v3 #30: rx' was not declared in this scope

GQster, 2024-06-26:
Trying to use the [LoRa_rx_tx](https://github.com/ropg/heltec_esp32_lora_v3/tree/main/examples/LoRa_rx_tx) example with platformIO and the 'regular board' but am getting this error:

```
src/main.cpp: In function 'void setup()':
src/main.cpp:55:23: error: 'rx' was not declared in this scope
   radio.setDio1Action(rx);
                       ^~
src/main.cpp: In function 'void loop()':
src/main.cpp:95:25: error: 'rx' was not declared in this scope
     radio.setDio1Action(rx);
```

My platformio.ini:

```
[env:heltec_wifi_kit_32_V3]
platform = espressif32
board = heltec_wifi_kit_32_V3
framework = arduino
lib_deps = ropg/Heltec_ESP32_LoRa_v3@^0.9.1

```

My only change to the code at this point is adding `#include <Arduino.h>` to the top.

I coppied the [LoRa_rx_tx.ino](https://github.com/ropg/heltec_esp32_lora_v3/blob/main/examples/LoRa_rx_tx/LoRa_rx_tx.ino) file into the src/main.cpp in the platformIO project

ropg, 2024-06-26:
In a .ino file, you can reference functions that are defined further down in the file, in a regular cpp file you cannot. Try lifting the definition of rx (all the way at the end) to before your main that calls it. Or have a declaration line at the beginning (outside of any functions) that just reads `void rx();`

GQster, 2024-06-27:
oh duh... Didn't see the rx() at the bottom. my bad!

closed 2024-06-27 by ropg