ropg/heltec_esp32_lora_v3 #34: no I2C

jo-ei, 2024-07-21:
Hello everyone,
I have been trying for days now to address an RTC clock in order to have the time locally after retrieval via LORA.

With the original lib from heltec and the arduino environment it works and I see the address with the I2C scanner.
With this lib it does not work and I always see only

> SDA on PIN: 41
> SCL on PIN: 42
> Scanning I2C Addresses Channel 1
> .. .. .. .. .. .. .. .. .. .. .. .. .. .. .. .. 
> .. .. .. .. .. .. .. .. .. .. .. .. .. .. .. .. 
> .. .. .. .. .. .. .. .. .. .. .. .. .. .. .. .. 
> .. .. .. .. .. .. .. .. .. .. .. .. .. .. .. .. 
> .. .. .. .. .. .. .. .. .. .. .. .. .. .. .. .. 
> .. .. .. .. .. .. .. .. .. .. .. .. .. .. .. .. 
> .. .. .. .. .. .. .. .. .. .. .. .. .. .. .. .. 
> .. .. .. .. .. .. .. .. .. .. .. .. .. .. .. .. 
> Scan Completed, 0 I2C Devices found.

is there a special way with this lib?

I have tried many things, here is the latest version of my test code:

```
#define SDA 41
#define SCL 42

void setup() {
  heltec_setup();
    Wire.begin(SDA, SCL);

  Serial.print("SDA on PIN: ");
  Serial.println(SDA);
  Serial.print("SCL on PIN: ");
  Serial.println(SCL);
}

void loop() {

  Serial.println("Scanning I2C Addresses Channel 1");
  uint8_t cnt=0;
  for(uint8_t i=0;i<128;i++){
    Wire.beginTransmission(i);
    uint8_t ec=Wire.endTransmission(true);
    if(ec==0){
      if(i<16)Serial.print('0');
      Serial.print(i,HEX);
      cnt++;
    }
    else Serial.print("..");
    Serial.print(' ');
    if ((i&0x0f)==0x0f)Serial.println();
    }
  Serial.print("Scan Completed, ");
  Serial.print(cnt);
  Serial.println(" I2C Devices found.");

delay(20000);

}

```
since i see the clock module with the other code, the wiring must be correct

can you give me a hint what i might be doing wrong?

Thanks

ropg, 2024-07-21:
I’ll try to have a look at this later this week.

jo-ei, 2024-08-11:
hi,
any news about the bug?
was it possible to reproduce?

ropg, 2024-08-12:
Hello there! I have been operated on my shoulder, and neec tio have my arm in a sling for another month. So doing anything with electronics and lots of wires is a lot harder now than I would like it to be. Two thoughts:

* Why not use the RTC in the ESP32? Does that need an extra crystal? (haven't used it with this board)
* Try putting `#define HELTEC_NO_DISPLAY` as first line to see if the I2C to display interferes.

Good luck!

jo-ei, 2024-08-13:
Hi ropg,
first of all, get well soon for your shoulder.

Unfortunately, the internal RTC does not continue to run during deep sleep and I don't want to get the times via LoRa every time unnecessarily. (or am I doing something wrong and there is a way to get the time even in deep sleep)

I have the light version without display.

I still set the option for testing and couldn't see any improvement.
Is there anything else I can do to provide more information?

Thank you

jo-ei, 2024-10-21:
Hello @decodeais,
thanks for your answer, but as the title from your code says, there are two I2C buses.
```
ESP32 have two I2C (I2C0 and I2C1) bus
* OLED is connected to I2C0, so if scan with Wire (I2C0), the return address should be 0x3C.
```
 
The one you described is only for the display.
Hence the following PIN configuration
```
static const uint8_t SCL_OLED = 15;
static const uint8_t SDA_OLED = 4;
```
But I have the model without display, so I can't test your part.

But since I have connected my RTC to Bus1, I used:
```
#define SDA 41
#define SCL 42
```

decodeais, 2024-10-22:
When I looked to the pinout of the my module  it looked like pin 8/9. The
numbers you use are so high.