ropg/heltec_esp32_lora_v3 #35: USB D+ and D- on headers

MichaelPWalter, 2024-07-22:
Hey I wanted to ask if there is a way to hardwire a simple USB connection over the headers ? I can supply 5v and ground over the Ve and GND pins. From the Pinout sheet found at (https://resource.heltec.cn/download/WiFi_Kit_32_V3/HTIT-WB32_V3.png) I saw that GPIO Pins 19 and 20 support USB D- and D+ respectively. However this does not seem to work for me. Also there is no direct electrical connection to the USB port on the Heltec V3 3.1.
Would love to know how to do this

MichaelPWalter, 2024-07-22:
I have made progress. The GPIO pins are different to what is shown in the schematic. Pin 19 and 20 are towards a corner of the board.

MichaelPWalter, 2024-07-22:
Connecting 5V directly to the 5V line, GND to GND and D- and D+ to GPIO PIN 19 and 20, Debian on the Raspberry recognizes the Heltec Lora 32 V3. 

Running `lsusb` results in `Bus 001 Device 022: ID 303a:1001 Espressif USB JTAG/serial debug unit` which from my undrestanding circumvents the CP2102 and uses the ESP 32 S3's built in UART. I can even connect to the device but never get a response. 

Running `sudo dmesg | grep tty` results in `[    0.000000] Kernel command line: coherent_pool=1M 8250.nr_uarts=0 snd_bcm2835.enable_headphones=0 snd_bcm2835.enable_hdmi=1 snd_bcm2835.enable_hdmi=0  smsc95xx.macaddr=B8:27:EB:FB:53:75 vc_mem.mem_base=0x1ec00000 vc_mem.mem_size=0x20000000  console=ttyS0,115200 console=tty1 root=PARTUUID=c546f6af-02 rootfstype=ext4 fsck.repair=yes rootwait cfg80211.ieee80211_regdom=US
[    0.000412] printk: console [tty1] enabled
[    2.893439] 3f201000.serial: ttyAMA1 at MMIO 0x3f201000 (irq = 99, base_baud = 0) is a PL011 rev2
[    2.898993] serial serial0: tty port ttyAMA1 registered
[   25.410866] systemd[1]: Created slice system-getty.slice - Slice /system/getty.
[   28.590884] cdc_acm 1-1:1.0: ttyACM0: USB ACM device
[  485.327434] cdc_acm 1-1:1.0: ttyACM0: USB ACM device
[  516.579059] cdc_acm 1-1:1.0: ttyACM0: USB ACM device
[  591.490785] cdc_acm 1-1:1.0: ttyACM0: USB ACM device
[  623.481389] cdc_acm 1-1:1.0: ttyACM0: USB ACM device
[  655.213561] cdc_acm 1-1:1.0: ttyACM0: USB ACM device
[  686.448040] cdc_acm 1-1:1.0: ttyACM0: USB ACM device
[  706.795151] cdc_acm 1-1:1.0: ttyACM0: USB ACM device
[  708.100039] cdc_acm 1-1:1.0: ttyACM0: USB ACM device
[  739.273195] cdc_acm 1-1:1.0: ttyACM0: USB ACM device
[  766.386859] cdc_acm 1-1:1.0: ttyACM0: USB ACM device
[  798.053382] cdc_acm 1-1:1.0: ttyACM0: USB ACM device
[  851.125712] cdc_acm 1-1:1.0: ttyACM0: USB ACM device
[  882.881321] cdc_acm 1-1:1.0: ttyACM0: USB ACM device
[  914.127543] cdc_acm 1-1:1.0: ttyACM0: USB ACM device
[  945.617545] cdc_acm 1-1:1.0: ttyACM0: USB ACM device
[  976.854854] cdc_acm 1-1:1.0: ttyACM0: USB ACM device
[ 1008.401882] cdc_acm 1-1:1.0: ttyACM0: USB ACM device
[ 1039.600008] cdc_acm 1-1:1.0: ttyACM0: USB ACM device
[ 1070.846846] cdc_acm 1-1:1.0: ttyACM0: USB ACM device
[ 1327.787355] cdc_acm 1-1:1.0: ttyACM0: USB ACM device`

The ESP 32 seems to be recoginzed at ttyACM0

running `meshtastic --noproto` sometimes gets a short response but no real serial info and seems to freeze i.e. no output. Upon terminating using CRTL + C results in `
^CTraceback (most recent call last):
  File "/home/walter/.local/bin/meshtastic", line 8, in <module>
    sys.exit(main())
             ^^^^^^
  File "/home/walter/.local/pipx/venvs/meshtastic/lib/python3.11/site-packages/meshtastic/__main__.py", line 1573, in main
    common()
  File "/home/walter/.local/pipx/venvs/meshtastic/lib/python3.11/site-packages/meshtastic/__main__.py", line 1064, in common
    client = meshtastic.serial_interface.SerialInterface(
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/walter/.local/pipx/venvs/meshtastic/lib/python3.11/site-packages/meshtastic/serial_interface.py", line 64, in __init__
    StreamInterface.__init__(
  File "/home/walter/.local/pipx/venvs/meshtastic/lib/python3.11/site-packages/meshtastic/stream_interface.py", line 50, in __init__
    self.connect()
  File "/home/walter/.local/pipx/venvs/meshtastic/lib/python3.11/site-packages/meshtastic/stream_interface.py", line 66, in connect
    self._writeBytes(p)
  File "/home/walter/.local/pipx/venvs/meshtastic/lib/python3.11/site-packages/meshtastic/stream_interface.py", line 92, in _writeBytes
    self.stream.flush()
  File "/home/walter/.local/pipx/venvs/meshtastic/lib/python3.11/site-packages/serial/serialposix.py", line 673, in flush
    termios.tcdrain(self.fd)
`

MichaelPWalter, 2024-07-22:
running meshtastic --noproto now resulted in 
`WARNING file:mesh_interface.py _sendToRadio line:788 Not sending packet because protocol use is disabled by noProto
Connected to radio
WARNING file:mesh_interface.py _sendPacket line:630 Not sending packet because protocol use is disabled by noProto
`
so to me it does connect but then it freezes or something

ropg, 2024-07-30:
I'm going to close this as it's not really an issue with my library. Do feel free to continue this under 'discussion'.

closed 2024-07-30 by ropg