ropg/heltec_esp32_lora_v3 #40: example LoRaWAN_TTN not working

BennoB666, 2024-08-19:
Hello Ropg,

Luckily I found your library. The Heltec libraries and instructions are terrible. I have a device on which I have installed the Lora example. After entering the TTN access data via the serial console I always get the following error message. What could be my mistake?

```
Enter LoRaWAN band (e.g. EU868 or US915)  [EU868]
Enter subband for your frequency plan, if applicable. Otherwise just press Enter.  []
Enter joinEUI (64 bits, 16 hex characters.) Press enter to use all zeroes.  [0000110220330440]
Enter devEUI (64 bits, 16 hex characters)  [70B3D57EDBD69B2C]
Enter appKey (128 bits, 32 hex characters)  [1B799FCFA9EABED1FE6863AC31237181]
Enter nwkKey (128 bits, 32 hex characters)  [1773074BA234B63D1683A13F76CCF6D1]
Thank you. Provisioning information saved to flash.
Now joining network.
Could not join network. We'll try again later.
Going to deep sleep now
Next TX in 900 s
```

BennoB666, 2024-08-19:
[image]

ropg, 2024-08-19:
The log you have included, and the image do not seem to fit together as the DevAddr does not match the devEUI. Can that be? Beyond that I don't know at first sight what the problem is. Why are you using this particular joinEUI?

BennoB666, 2024-08-19:
DeviceAdr. is 260B2A97 in TTN

BennoB666, 2024-08-19:
The devEUI a changed before i post.

BennoB666, 2024-08-19:
I tried JoinEUI all zero also. Same result.

FlorianHeigl, 2024-09-07:
@BennoB666 did you find anything more? same issue here, or worse. I didn't manage to get as far as seeing messages in TTN any more after switching to this library. 
(I can't recall specifics anymore, most of today I lost figuring out I needed to use Lora version 1.1 over the 1.0.2 from Heltec's docs)

ropg, 2024-09-07:
Are you sure you are close enough to a working gateway? Does other software (on same board if possible) get a connection?

BennoB666, 2024-09-07:
1 m away from the gateway. With the Heltec library and same setup it connects immediately. For now i go back to the Heltec Library.

FlorianHeigl, 2024-09-10:
i was able to reach the gw with the heltec stack - though that was using Lora 1.0.2. I don't understand enough of the stuff yet to know what matters and what doesn't.
But just wanted to say I'm interested to help make it work, I'll come back in a month-ish after your shoulder is hopefully already forgotten.

ropg, 2024-09-11:
I'm guessing this may be a protocol issue of some kind at/behind your gateway. I haven't really played with LoRaWAN prior to 1.1.0, and I don't know what, if anything, changes on the network side by selecting one or the other when creating a node at the TTN side.

carlosmdarribas, 2024-10-11:
Hello! Same issue here, and everything you guys said is also checked... are there any way to debug? It just keeps trying to connect in loop.

[image]

ropg, 2024-10-11:
Eh, just noticed you have "reset join nonces" on at TTN. That should not be needed (but should still not be causing this immediate rejoin.)

If you are sure the board is not resetting for some hardware/power reason, maybe describe your entire setup: what version of all the relevant libraries? (this library, RadioLib, LoRaWAN_ESP32), photo of all hardware, and if possible a screen movie (phone video is fine) showing serial entry of data from new TTN account while TTN log for that account is visible in other window.

carlosmdarribas, 2024-10-11:
First of all, thank you for your quick response! Yes, using the Wireless Stick V3 of Heltec.

I've created a new one to match yours in case I misconfigured something:
[image]

And the source code is just the same, just added the line "#define HELTEC_WIRELESS_STICK":
[image]

The quick ones that happened was my fault, I modified not to wait 900s just 4

carlosmdarribas, 2024-10-11:
Regarding versions:

- LoRaWAN ESP32: v1.20
- This lib Heltec_ESP32_LoRa_v3: v0.9.2
- RadioLib: v7.0.2

Regarding hardware setup, is just this board connected via USB C to my mac. The gateway is a Mikrotik LR8 Kit connected to a self-hosted TTS... Everything out-of-the-box

ropg, 2024-10-11:
And what does the TTN side show? Can you test by using an application on TTNs sandbox just to make everything as much the same as my environment as possible?

ropg, 2024-10-11:
Also you used all zeroes as joinEUI on the device and D8 .. FD on the other end, it seems.

carlosmdarribas, 2024-10-11:
Moving to TTN sandbox... meanwhile it was just an old screenshow :D that's now
[image]

ropg, 2024-10-11:
I haven't tested TTN with a stick, I think. Will do that to see if that's the difference if this doesn't solve it.

carlosmdarribas, 2024-10-11:
I'm sorry about that, my fault. 

Now I've duplicated the values on the gateway (now points to 2 servers: on premise and TTN EU), and the same config (including Rev A) on both ones. Same behavior. I receive the "ping" but empty data on both:

[image]

And yes, as you said Arduino IDE for MacOS also v2.3.3. This is quite a mistery! (Again, thanks for your help)

ropg, 2024-10-11:
Can I see the whole log from TTN?

carlosmdarribas, 2024-10-11:
of course, this is:

[image]

carlosmdarribas, 2024-10-11:
with verbose option, if it helps:
[image]

ropg, 2024-10-11:
Somehow the join acknowledgement is not making it back to the endpoint. We're gonna need the RadioLib debug output. Modify /src/BuildOptUser.h in the RadioLib library directory and uncomment this line:

`//#define RADIOLIB_DEBUG_PROTOCOL`

Then try again and send output

ropg, 2024-10-11:
(I'm also not sure about this business with "now points to 2 servers: on premise and TTN EU". Mever ran my own, no idea if that's not somehow causing this.)

ropg, 2024-10-11:
(You'll notice my LoRaWAN_ESP32 lib adds its own output in the debug output, prefixed with `[persist]`)

carlosmdarribas, 2024-10-11:
I modified but it did not show any more output (recompiled, rerun, cleared flash...), but I think I am not seeing logs, as I don't see any log starting with this prefix... maybe am I missing something?

these are the ones I see: 

[image]

ropg, 2024-10-11:
That temperature seems a bit on the high side, and RadioLib should **definitely** be producing much more output with `RADIOLIB_DEBUG_PROTOCOL` turned on. Start over with a new endpoint on TTN, clear the flash so that you can enter that new endpoint.

If RadioLib is really not producing more output, you may be using (some?) libraries from a different directory than you think, or something crazy like that.

Anyway, I gave you some pointers to experiment, I'm off to bed. Will see in the morning where you landed.

Hasta luego y buenas noches

carlosmdarribas, 2024-10-11:
I'll be reinstalling all the stack until I get more output... I'll keep you updated.

Guten Abend, und vielen Dank!

carlosmdarribas, 2024-10-12:
here again! after reinstalling everything and trying to use directly RadioLib library (with no success, looks like something about keys, not important to me) now I've got working logs where I see some interesting things:

This log message:
```
RLB_PRO: [persist] Nonces and session data restored from RTC RAM
RLB_PRO: The Session buffer (0000) does not match the Nonces buffer (09e1)
```

After that, this is the moment when it fails:

```
LB_PRO: PHY:  Frequency DL = 869.525 MHz
RLB_PRO: LoRa: SF = 12, TX = 16 dBm, BW = 125.0 kHz, CR = 4/5
RLB_PRO: Opening Rx2 window (688 ms timeout)... <-- Rx Delay end 
RLB_PRO: Closing Rx2 window
RLB_PRO: [persist] Nonces and session data saved to RTC RAM
RLB_PRO: [persist] Nonces saved to NVS. (Only actually written if changed.)
Could not join network. We'll try again later.
Going to deep sleep now
RLB_PRO: [persist] Nonces and session data saved to RTC RAM
RLB_PRO: [persist] Nonces saved to NVS. (Only actually written if changed.)
```

does it mean anything for you?

ropg, 2024-10-12:
`Session buffer [...] does not match the Nonces buffer` is a checksum mechanism used by the LoRaWAN code in RadioLib to see if the two kinds of data it stores long-term describe the same state. The 0000 seems to indicate to me that somehow the RTC RAM, where my code stores the session state, is not kept.

Do you have the Stick (with little screen), or the Stick Lite (no screen) ?

ropg, 2024-10-12:
And do you have a battery connected?

carlosmdarribas, 2024-10-12:
The Wireless Stick with display, even though I've got it software disabled. No battery, just connected to my mac via its USB C port

ropg, 2024-10-12:
I've replicated your situation as closely as I know how to and I must say it works for me.

* Helteck Stick v3, no battery, connected to Mac via USB
* Mac running Arduino IDE version 2.3.3
* Heltec_ESP32_LoRa_v3 version 1.0.3
* RadioLib version 7.0.2
* LoRaWAN_ESP32 version 1.2.0
* Compiled the LoRaWAN_TTN.ino example with added `#define HELTEC_NO_DISPLAY` as only modification.
* Uncommented `RADIOLIB_DEBUG_PROTOCOL` in RadioLib.
* Set "Tools / Erase all flash before sketch upload" to make sure there's no old provisioning data hanging around.

* Created a new end device 'heltec-stick' on TTN:

[image]

* Copy-paste the info into the serial dialog and away we go:

```
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x28 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3818,len:0x109c
load:0x403c9700,len:0x4
load:0x403c9704,len:0xb50
load:0x403cc700,len:0x2fe4
entry 0x403c98ac
Temperature: 27.80 °C
Radio init
RLB_PRO: [persist] Reading from NVS
RLB_PRO: [persist] No or incomplete provisioning. Getting from console.
No provisioning data found in flash.
Please enter the provisioning information needed to join the LoRaWAN network.

Enter LoRaWAN band (e.g. EU868 or US915)  [EU868]
Enter subband for your frequency plan, if applicable. Otherwise just press Enter.  []
Enter joinEUI (64 bits, 16 hex characters.) Press enter to use all zeroes.  [0000000000000000]
Enter devEUI (64 bits, 16 hex characters)  [70B3D57ED006B0A0]
Enter appKey (128 bits, 32 hex characters)  [51322616E7E6F0A1C0FAB578AFA77E16]
Enter nwkKey (128 bits, 32 hex characters)  [E698A10DEAA159A4CF57121C06F68BD0]
RLB_PRO: LoRa: SF = 10, TX = 16 dBm, BW = 125.0 kHz, CR = 4/5
RLB_PRO: Opening Rx1 window (231 ms timeout)... <-- Rx Delay end 
RLB_PRO: Closing Rx1 window
RLB_PRO: JoinAccept (JoinNonce = 1, previously 0):
RLB_PRO: 00000000: 20 01 00 00 13 00 00 75 08 0b 26 83 05 18 4f 84   ......u..&...O.
RLB_PRO: 00000010: e8 56 84 b8 5e 84 88 66 84 58 6e 84 00 48 be c4  .V..^..f.Xn..H..
RLB_PRO: 00000020: 16                                               .                 
RLB_PRO: LoRaWAN revision: 1.1
RLB_PRO: Setting up dynamic channels
RLB_PRO: UL:   0 1 868.100 (0 - 5) | DL:   0 1 868.100 (0 - 5)
RLB_PRO: UL:   1 1 868.300 (0 - 5) | DL:   1 1 868.300 (0 - 5)
RLB_PRO: UL:   2 1 868.500 (0 - 5) | DL:   2 1 868.500 (0 - 5)
RLB_PRO: [MAC] 0x05
RLB_PRO: 00000000: 03 d2 ad 84                                      ....              
RLB_PRO: RXParamSetupReq: Rx1DrOffset = 0, rx2DataRate = 3, freq = 869.525
RLB_PRO: [MAC] 0x08
RLB_PRO: 00000000: 05                                               .                 
RLB_PRO: RXTimingSetupReq: delay = 5 sec
RLB_PRO: Processing CFList
RLB_PRO: [MAC] 0x07
RLB_PRO: 00000000: 03 18 4f 84 50                                   ..O.P             
RLB_PRO: UL:   3 1 867.100 (0 - 5) | DL:   3 1 867.100 (0 - 5)
RLB_PRO: [MAC] 0x07
RLB_PRO: 00000000: 04 e8 56 84 50                                   ..V.P             
RLB_PRO: UL:   4 1 867.300 (0 - 5) | DL:   4 1 867.300 (0 - 5)
RLB_PRO: [MAC] 0x07
RLB_PRO: 00000000: 05 b8 5e 84 50                                   ..^.P             
RLB_PRO: UL:   5 1 867.500 (0 - 5) | DL:   5 1 867.500 (0 - 5)
RLB_PRO: [MAC] 0x07
[... 31 lines cut ...]
RLB_PRO: 00000000: 41 ff 00 00 00 00 00 00 00 00 00 00 00 01        A.............    
RLB_PRO: LinkAdrReq: dataRate = 4, txSteps = 1, nbTrans = 1
RLB_PRO: UL:   0 1 868.100 (0 - 5) | DL:   0 1 868.100 (0 - 5)
RLB_PRO: UL:   1 1 868.300 (0 - 5) | DL:   1 1 868.300 (0 - 5)
RLB_PRO: UL:   2 1 868.500 (0 - 5) | DL:   2 1 868.500 (0 - 5)
RLB_PRO: UL:   3 1 867.100 (0 - 5) | DL:   3 1 867.100 (0 - 5)
RLB_PRO: UL:   4 1 867.300 (0 - 5) | DL:   4 1 867.300 (0 - 5)
RLB_PRO: UL:   5 1 867.500 (0 - 5) | DL:   5 1 867.500 (0 - 5)
RLB_PRO: UL:   6 1 867.700 (0 - 5) | DL:   6 1 867.700 (0 - 5)
RLB_PRO: UL:   7 1 867.900 (0 - 5) | DL:   7 1 867.900 (0 - 5)
RLB_PRO: LinkAdrAns: 07
RLB_PRO: [MAC] 0x06
RLB_PRO: DevStatusAns: status = 0xff08
Message sent, downlink received.
Going to deep sleep now
RLB_PRO: [persist] Nonces and session data saved to RTC RAM
RLB_PRO: [persist] Nonces saved to NVS. (Only actually written if changed.)
Next TX in 1194 s
```

Which makes my data show up at TTN.

I have seriously no clue what could be different on your end at this point, but at least now you have a complete working situation to compare against.

Good luck!

ropg, 2024-10-12:
(Note that copy-pasting from the Arduino interface is harder than it needs to be, so I just updated the paste so it no longer misses a chuck in the middle)

askmurphy, 2024-11-01:
I have the same issue with my Heltec Wireless stick..  
Serial Output:
```
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x9 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3818,len:0x109c
load:0x403c9700,len:0x4
load:0x403c9704,len:0xb50
load:0x403cc700,len:0x2fe4
entry 0x403c98ac
Temperature: 77.3 °C
Radio init
Could not join network. We'll try again later.
Going to deep sleep now
```

ropg, 2024-11-01:
Oddly enough all the reports for non-working connections have rather high temperatures, as measured by the ESP32 internally. If you power it down for the night, does it come up with a temp like this?

carlosmdarribas, 2024-11-01:
In my case (I still haven't made it work) it didn't matter the uptime. I bought other modules (the lite v3, the LoRa v2) with identical results, and even when they were just connected after some time without being used, they still worked the same.

askmurphy, 2024-11-01:
> In my case (I still haven't made it work) it didn't matter the uptime. I bought other modules (the lite v3, the LoRa v2) with identical results, and even when they were just connected after some [...]

Have you re-compiled it again ?