ropg/heltec_esp32_lora_v3 #41: Adoption of the Vision Master E290

Genaker, 2024-08-24:
I made adoption of your code with the Vision Master E290

https://heltec.org/project/vision-master-e290/

It requires no code just to #define some variables. However, it is better to add some logic connected with this device. It runs with No display mode and also with the specific define flag. 

Would you like to explore the opportunity to merge these changes to the code and add support for all e-ink devices?

Glenno-H, 2024-08-25:
Ive got a E290, and would love to see that!

ropg, 2024-08-25:
Oh, that is a nice board!

I'm a bit stuck at the moment, being behind on some other work and only able to use my left hand because of a shoulder operation. I'm interested, and I suggest forking this repository and making all the changes necessary to support this device. 

It would be nice if there was some automatic way of detecting which device is hooked up.

Genaker, 2024-09-12:
> Oh, that is a nice board!
>
> I'm a bit stuck at the moment, being behind on some other work and only able to use my left hand because of a shoulder operation. I'm interested, and I suggest forking [...]

You have an option to detect which device is hooked up. To do so, add a build parameter to platformio.io with the board's name. 
```
[env:vision-master-e290]
platform = espressif32
board = heltec_wifi_lora_32_V3
framework = arduino
monitor_speed = 115200
monitor_filters = esp32_exception_decoder
board_upload.use_1200bps_touch = true
build_flags = 
	-DHELTEC_BOARD=37
	-DE290
	-DESP32
	-DVision_Master_E290
```

There is a solution how to load your library with E290 and T 190:

```
// Disabling default Heltec lib OLED display
#define HELTEC_NO_DISPLAY
// Redefine button pin 
#define BUTTON 21
// Without this line, Lora Radio doesn't work with Heltec lib
#define ARDUINO_heltec_wifi_32_lora_V3
#include "heltec_unofficial.h"
```

Genaker, 2024-09-12:
I tested your library with T290 e-ink and T190 TFT display. It works

kaspesla, 2024-09-27:
Did you actually get the screen working on the T190?

Genaker, 2024-09-27:
Yes. The oh need custom libraries. This library run without display

kaspesla, 2024-09-27:
Thanks. What libraries work with this display? I'm having no luck finding anything. It's not even mentioned in Heltec's documentation!

Genaker, 2024-09-27:
There it is: 

'''

// #define LoRaWAN_DEBUG_LEVEL 0
#include "HT_ST7789spi.h"
#include "global_config.h"
#include "images.h"
#include "ui.h"
#include <Adafruit_GFX.h>
#include <Arduino.h>

#define st7789_CS_Pin 39
#define st7789_REST_Pin 40
#define st7789_DC_Pin 47
#define st7789_SCLK_Pin 38
#define st7789_MOSI_Pin 48
#define st7789_LED_K_Pin 17
#define st7789_VTFT_CTRL_Pin 7

// lcd object pointer, it's a 240x135 lcd display, Adafruit dependcy
static HT_ST7789 *st7789 = NULL;
static SPIClass *gspi_lcd = NULL;

char buffer[256];

// Disabling default Heltec lib OLED display
#define HELTEC_NO_DISPLAY
#define DISPLAY_WIDTH 320
#define DISPLAY_HEIGHT 170
// Without this line Lora Radio doesn't work with heltec lib
#define ARDUINO_heltec_wifi_32_lora_V3
// T190 button pin
#define BUTTON GPIO_NUM_21
#define HELTEC_POWER_BUTTON
#include "heltec_unofficial.h"

'''

kaspesla, 2024-09-27:
Thank you! I will try it tomorrow.

jamesgbahr, 2024-12-07:
Trying everything to get this working for t190 board but no luck where are the UI. And global.h files coming from. Every thing I try in Arduino IDE says cannot compile for t190 board.