ropg/heltec_esp32_lora_v3 #42: Heltec WiFi LoRa 32(V3) / Wireless shell(V3) not in the list of boards

probert100, 2024-09-02:
Did I miss something if I don't see "Heltec WiFi LoRa 32(V3) / Wireless shell(V3)" in the list of the boards?
I see "Heltec WiFi LoRa 32(V3)"  

[image]

I believe I followed all of the instructions, but any example I try has this error: `Error: esp_usb_jtag: could not find or open device!`  Get Board Info returns basic information and when I plug it/unplug it. Serial Monitor shows state changes.

probert100, 2024-09-03:
I was able to figure out what my problem was.

closed 2024-09-03 by probert100

KonRoz, 2024-09-05:
Which board did you end up using? I don't see "Heltec WiFi LoRa 32(V3) / Wireless shell(V3)" either.