ropg/heltec_esp32_lora_v3 #46: V3.1 Battery volume

saber2r, 2024-09-23:
Hi 

I have 5 heltec V3.1 Dev-Boards

3 hletec WiFi Lora 32 V3.1
2 heltec Wireless stick lite V3.1

Only 1 Lora 32 V3.1 can show to 4.26 100% others stoped on 4.10 95% ...(1000mah and 700mah capacity as same)

I'm a newbie to Arduino, so if anyone have any idea that would be grateful..

` 
    char bufbat[80];

    if (!last_print || millis() - last_print > INTERVAL * 1000) {

    float vbat = heltec_vbat(); 
    Serial.printf("Battery : %.2f V (%d%%)\n", vbat, heltec_battery_percent(vbat)); 
    sprintf(bufbat,"%.2fV ( %d%% )", vbat, heltec_battery_percent(vbat));    
    last_print = millis();

  }`

ropg, 2024-09-24:
The battery curve I used to calculate percentage from voltage was made by a program that is included with the repository. The following lines are from heltec_unoffical.h:

```cpp
// Don't you just hate it when battery percentages are wrong?
//
// I measured the actual voltage drop on a LiPo battery and these are the
// average voltages, expressed in 1/256'th steps between min_voltage and
// max_voltage for each 1/100 of the time it took to discharge the battery. The
// code for a telnet server that outputs battery voltage as CSV data is in
// examples, and a python script that outputs the constants below is in
// src/tools.
const float min_voltage = 3.04;
const float max_voltage = 4.26;
const uint8_t scaled_voltage[100] = {
  254, 242, 230, 227, 223, 219, 215, 213, 210, 207,
  206, 202, 202, 200, 200, 199, 198, 198, 196, 196,
  195, 195, 194, 192, 191, 188, 187, 185, 185, 185,
  183, 182, 180, 179, 178, 175, 175, 174, 172, 171,
  170, 169, 168, 166, 166, 165, 165, 164, 161, 161,
  159, 158, 158, 157, 156, 155, 151, 148, 147, 145,
  143, 142, 140, 140, 136, 132, 130, 130, 129, 126,
  125, 124, 121, 120, 118, 116, 115, 114, 112, 112,
  110, 110, 108, 106, 106, 104, 102, 101, 99, 97,
  94, 90, 81, 80, 76, 73, 66, 52, 32, 7,
};
```

To calibrate for any other battery, run the [`battery_discharge_telnet`](https://github.com/ropg/heltec_esp32_lora_v3/tree/main/examples/battery_discharge_telnet) example, connect via WiFi and telnet (bc you can't drain the battery when connected via USB-C), log the CSV data to file, give that file as argument to [`src/tools/battery_constants.py`](https://github.com/ropg/heltec_esp32_lora_v3/blob/main/src/tools/battery_constants.py) and you'll get the same set of lines to insert in the .h file instead of what's in there now.

saber2r, 2024-10-07:
> The battery curve I used to calculate percentage from voltage was made by a program that is included with the repository. The following lines are from heltec_unoffical.h:
> 
> ```c++
> // Don't you just hate it when battery percentages are wrong?
> //
> // I measured the actual voltage drop on a LiPo battery and these are the
> // average voltages, expressed in 1/256'th steps between min_voltage and
> // max_voltage for each 1/100 of the time it took to discharge the battery. The
> // code for a telnet server that outputs battery voltage as CSV data is in
> // examples, and a python script that outputs the constants below is in
> // src/tools.
> const float min_voltage = 3.04;
> const float max_voltage = 4.26;
> const uint8_t scaled_voltage[100] = {
>   254, 242, 230, 227, 223, 219, 215, 213, 210, 207,
>   206, 202, 202, 200, 200, 199, 198, 198, 196, 196,
>   195, 195, 194, 192, 191, 188, 187, 185, 185, 185,
>   183, 182, 180, 179, 178, 175, 175, 174, 172, 171,
>   170, 169, 168, 166, 166, 165, 165, 164, 161, 161,
>   159, 158, 158, 157, 156, 155, 151, 148, 147, 145,
>   143, 142, 140, 140, 136, 132, 130, 130, 129, 126,
>   125, 124, 121, 120, 118, 116, 115, 114, 112, 112,
>   110, 110, 108, 106, 106, 104, 102, 101, 99, 97,
>   94, 90, 81, 80, 76, 73, 66, 52, 32, 7,
> };
> ```
> 
> To calibrate for any other battery, run the [`battery_discharge_telnet`](https://github.com/ropg/heltec_esp32_lora_v3/tree/main/examples/battery_discharge_telnet) example, connect via WiFi and telnet (bc you can't drain the battery when connected via USB-C), log the CSV data to file, give that file as argument to [`src/tools/battery_constants.py`](https://github.com/ropg/heltec_esp32_lora_v3/blob/main/src/tools/battery_constants.py) and you'll get the same set of lines to insert in the .h file instead of what's in there now.

I did two different capacity batteries curve ,1000mah and 700mah.
But still can't get correct volume of battery volume, so I change the parameter at row 208 in heltec_unoffical.h

`float vbat = analogRead(VBAT_ADC) / 238.7;`

If showing 4.10 V when the battery is fully charged.

And do a simple calculation :

(238.7 X 4.1V) / 4.2V =  233.016 

so I change 238.7 to 233.016 to match real ADC read value

[1000mAh_discharge2.csv](https://github.com/user-attachments/files/17281438/1000mAh_discharge2.csv)

PS : This CSV data that I made  from HELTEC 1000mah High temperature Version battery ,for share.