ropg/heltec_esp32_lora_v3 #47: LoRaWAN_ESP32.h: No such file or directory

m2mlorawan, 2024-09-25:
C:\Users\User\AppData\Local\Temp\arduino\sketches\B5C045929F5B318410FCADB1B6E80B2A\sketch\test.ino.cpp -o nul
Alternatives for LoRaWAN_ESP32.h: []
ResolveLibrary(LoRaWAN_ESP32.h)
  -> candidates: []
C:\Users\User\Downloads\test\test.ino:24:10: fatal error: LoRaWAN_ESP32.h: No such file or directory
 #include <LoRaWAN_ESP32.h>
          ^~~~~~~~~~~~~~~~~
compilation terminated.
exit status 1

Compilation error: LoRaWAN_ESP32.h: No such file or directory

decodeais, 2024-09-29:
I think you forgot to install the Library LoRaWan__ESP32. Open the library  manager install it and then it should work.

m2mlorawan, 2024-09-30:
Thank.
Without doing anything, try to compile again and it passed that error but get struck at 

C:\Users\User\Downloads\heltec_esp32_lora_v3-main\heltec_esp32_lora_v3-main\examples\LoRaWAN_TTN\LoRaWAN_TTN.ino:67:45: error: 'RADIOLIB_LORAWAN_NO_DOWNLINK' was not declared in this scope
   if(state == RADIOLIB_ERR_NONE || state == RADIOLIB_LORAWAN_NO_DOWNLINK) {
                                             ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
C:\Users\User\Downloads\heltec_esp32_lora_v3-main\heltec_esp32_lora_v3-main\examples\LoRaWAN_TTN\LoRaWAN_TTN.ino:67:45: note: suggested alternative: 'RADIOLIB_LORAWAN_DOWNLINK'
   if(state == RADIOLIB_ERR_NONE || state == RADIOLIB_LORAWAN_NO_DOWNLINK) {
                                             ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
                                             RADIOLIB_LORAWAN_DOWNLINK
exit status 1

Compilation error: 'RADIOLIB_LORAWAN_NO_DOWNLINK' was not declared in this scope

decodeais, 2024-09-30:
Look at issue #51 , I have the same problem.

ropg, 2024-09-30:
LoRaWAN_TTN example now works with latest RadioLib ( >= 7.0.0)

closed 2024-09-30 by ropg