ropg/heltec_esp32_lora_v3 #51: Where is RADIOLIB_LORAWAN_NO_DOWNLINK ?

decodeais, 2024-09-29:
**Setup:**
- **Board**: Heltec WiFi LoRa 32 (V3)
- **Library version**: LoRaWan_ESP v1.1.0
- **Heltec_ESP32_LoRa v0.9.1
- **RadioLib  v7.0.0
- ** ESP8266 and ESP32  OLED Driver ... v4.6.1

The example throws a compilation error related to `RADIOLIB_LORAWAN_NO_DOWNLINK` not being declared.

I have no idea what might be causing this issue or how to resolve it. Any guidance or suggestions would be appreciated.
 
Thank you!

m2mlorawan, 2024-09-30:
Happen when I compile LoRaWAN_TTN.ino in examples dir.

decodeais, 2024-09-30:
Sorry I forgot to tell what I am compiling. 
It was also LoRaWAN_TTN.ino.

decodeais, 2024-09-30:
I found the solution. We have to downgrade RadioLib from version 7.0.0 to 6.6.0 .
What is wrong ? Is it a bug of LoRaWan_TT or is the bug in RadioLib

decodeais, 2024-09-30:
> Ich habe die Lösung gefunden. Wir müssen RadioLib von Version 7.0.0 auf 6.6.0 downgraden. Was ist falsch? Ist es ein Fehler von LoRaWan_TT oder ist der Fehler in RadioLib

ropg, 2024-09-30:
There are a number of issues, both in this library and in my [LoRaWAN_ESP32](https://github.com/ropg/LoRaWAN_ESP32) library that have to do with incompatibilities introduced by RadioLib 7.0.0. Hoping to solve all these in the course of today.

decodeais, 2024-09-30:
With Version 6.6 it runs. I think Heltec will search for a solution forthis problem. My problem is that I have no instruction how to use, where I get the ID's ....

decodeais, 2024-09-30:
Now the workaround gets a little more complicate:
Lorawan_ESP32 has to be downgraded to v1.1.0
RadiloLib  has to be downgraded to v6.6.0

ropg, 2024-09-30:
If you use the example in LoRaWAN_ESP32, you should be able to use the current versions of both RadioLib(7.0.1) and LoRaWAN_ESP32 (1.2.0). Later tonight, you can use the example from the current version of this library too.

ropg, 2024-09-30:
LoRaWAN_TTN example now works with latest RadioLib ( >= 7.0.0)

closed 2024-09-30 by ropg

decodeais, 2024-09-30:
Thank you !!
 I will test it as soon possible.