ropg/heltec_esp32_lora_v3 #60: PlatformIO is much better than the Arduino IDE.

decodeais, 2024-10-21:
PlatformIO is much better than the Arduino IDE. 
1st argument for me is the compile + upload time. 
If you change one line code would take Arduino IDE: 1:50 min; PlatformIO: 0:25

How To:
it is very easy from the PIO Home -> Import Arduino Project
Select the board: Heltec WiFi LoRa 32 (V3)
select one *.ino file from the examples

The platfomio.ini looks then like this:

```
[env:heltec_wifi_kit_32_V3]
platform = espressif32
board = heltec_wifi_kit_32_V3
framework = arduino
lib_deps = ropg/Heltec_ESP32_LoRa_v3@^0.9.1
```

I you create a new PlatformIO project from the wizard
Select the board: Heltec WiFi LoRa 32 (V3)
Select Framework: Arduino
Add the library ropg/Heltec_ESP32_LoRa_v3 to the project via PIO Home -> Libraries or just add it to the platformio.ini as 
```
lib_deps = ropg/Heltec_ESP32_LoRa_v3@^0.9.1
```

Copy all from the example *.ino to the main.cpp (if you have a main.c rename it to main.cpp)

If you facing errors like this:

.pio/libdeps/heltec_wifi_lora_32_V3/Heltec_ESP32_LoRa_v3/src/heltec_unofficial.h:33:19: error: 'const uint8_t GPIO_NUM_8' redeclared as different kind of symbol
 #define SS        GPIO_NUM_8

The change all these lines in the .pio/libdeps/heltec_wifi_lora_32_V3/Heltec_ESP32_LoRa_v3/src/heltec_unofficial.h
from
```
// SPI pins
#define SS        GPIO_NUM_8
#define MOSI      GPIO_NUM_10
#define MISO      GPIO_NUM_11
#define SCK       GPIO_NUM_9
// Radio pins
#define DIO1      GPIO_NUM_14
#define RST_LoRa  GPIO_NUM_12
#define BUSY_LoRa GPIO_NUM_13
// Display pins
#define SDA_OLED  GPIO_NUM_17
#define SCL_OLED  GPIO_NUM_18
#define RST_OLED  GPIO_NUM_21
```
to:
```
// SPI pins
//#define SS        GPIO_NUM_8
//#define MOSI      GPIO_NUM_10
//#define MISO      GPIO_NUM_11
//#define SCK       GPIO_NUM_9
// Radio pins
#define DIO1      GPIO_NUM_14
//#define RST_LoRa  GPIO_NUM_12
//#define BUSY_LoRa GPIO_NUM_13
// Display pins
//#define SDA_OLED  GPIO_NUM_17
//#define SCL_OLED  GPIO_NUM_18
//#define RST_OLED  GPIO_NUM_21
```

 This is necessary due to the fact that these are already defined in the 
.platformio/packages/framework-arduinoespressif32/variants/heltec_wifi_lora_32_V3/pins_arduino.h

If you face issues like this:

```
src/main.cpp: In function 'void setup()':
src/main.cpp:53:23: error: 'rx' was not declared in this scope
   radio.setDio1Action(rx);
                       ^~
src/main.cpp: In function 'void loop()':
src/main.cpp:95:25: error: 'rx' was not declared in this scope
     radio.setDio1Action(rx);
                         ^~
src/main.cpp:111:29: error: 'rx' was not declared in this scope
         radio.setDio1Action(rx);
```
then just move the function 
```
void rx() {
  rxFlag = true;
}
```
higher or even on top of the file before the first usage.

_Originally posted by @knowthelist in https://github.com/ropg/heltec_esp32_lora_v3/issues/6#issuecomment-2381591775_

zombodotcom, 2024-10-23:
i still have the same 

```
.pio/libdeps/heltec_wifi_kit_32_V3/Heltec_ESP32_LoRa_v3/src/heltec_unofficial.h:44:19: error: 'const uint8_t GPIO_NUM_17' redeclared as different kind of symbol
 #define SDA_OLED  GPIO_NUM_17
 ```
 
 issues. with changing the heltec_unoffical.h
 
 
 
[image]

zombodotcom, 2024-10-23:
I deleted the library

and reinstalled

[image]

```
#define RST_LoRa GPIO_NUM_12
#define BUSY_LoRa GPIO_NUM_13
```

I used 12 and 13 because that's what it is in the pin def

[image]

platformio.ini
```
[env:heltec_wifi_kit_32_V3]
platform = espressif32
board = heltec_wifi_kit_32_V3
framework = arduino
lib_deps = ropg/Heltec_ESP32_LoRa_v3@^0.9.2
monitor_speed = 115200
```
Works

[image]

EmbeddedDevver, 2025-03-22:
I just came here to say THANKYOU for posting this!