ropg/heltec_esp32_lora_v3 #62: Not receiving join-accept

WhiptongueHydra, 2024-10-28:
Hi, sorry to bother you. Was just wondering what regional parameters you were using? I am setting up the node with the following params:
[image]
I have tried every type of regional parameters but the heltec still is unable to receive the join-accept or any downlinks:
[image]
Just want to be able to rule this out as an issue, could not find which ones you were using in any of the documentation. Thanks :)

WhiptongueHydra, 2024-10-28:
I am still having same issues unfortunately, I am not sure why. It will not receive downlinks.

ropg, 2024-10-28:
Yeah, you probably/hopefully found [this issue](https://github.com/ropg/heltec_esp32_lora_v3/issues/40), which tells you everything I do, in the greatest level of detail, to get a working setup. If you're doing exactly the same thing and it doesn't work, I'm really not sure how to help further. Please do post if you figure it out.