ropg/heltec_esp32_lora_v3 #63: Diaplay not Work

Rumhocker, 2024-10-30:
Hello, can anyone here confirm that the display with the examples on the Heltech LoRa V3.2 board does not work? Or am I too stupid?

ropg, 2024-10-31:
They may or may not have put the display on the "external" power like they did with the stick. Can you try running the display with `heltec_ve(true)` after `heltec_setup()` ?

Rumhocker, 2024-10-31:
Yes, this change my day. Display now show something. Thank you.

[Schematic_Diagram LoRa V3.2](https://resource.heltec.cn/download/WiFi_LoRa_32_V3/WiFi_LoRa_32_V3.2_Schematic_Diagram.pdf)

Rumhocker, 2024-10-31:
This is my Display and Button test:
`#define HELTEC_POWER_BUTTON   
#include <heltec_unofficial.h>

HotButton extbutton(7, true, LOW); //GPIO 7 without true and LOW dont´t work!!!

void setup() {
  Serial.begin(115200);
  heltec_setup();
  heltec_ve(true);
}

void loop() {
  heltec_loop();

  button.update();
  if (button.isSingleClick()) {
    display.println("The button was clicked.");
    heltec_led(30);
  }
  extbutton.update(); //extern button
  if (extbutton.isSingleClick()) {
    display.println("The Extern was clicked.");
    heltec_led(0);
  }
}`

MarcoTerzoloGit, 2024-11-01:
Can confirm that display was dead without chances to turn on before heltec_ve(true); Thank you for both question and answer!

Rumhocker, 2024-12-02:
Please set your serial monitor to 9600. Then we see what's happen.

jamesgbahr, 2024-12-04:
i had gotten it to work then updated libraries for t190 and now the same code that worked, is no longer working what exact setup do you use for board managers version installed and the json in preferences... i dont understand why my working code is no loger working and none of the demo code is displaying on oled

Kaubisch, 2024-12-04:
Thanks for reply!
I also tried to include Serial.begin(9600) and serial monitor is set to 9600 too.
The board manager was set to the recommended link. I'll send you later to double confirm.
After the trials I reinstalled Meshtastic and everything works fine.