ropg/heltec_esp32_lora_v3 #67: Battery voltage at V3.2 Boards

Rumhocker, 2024-11-21:
The heltec_vbat() return 0.00 at v3.2 boards. After searching around I found out, that the logic of ADC_Ctrl is changed from LOW enable to HIGH enable.
[Heltec LoRa32 v3.2 charge status](https://github.com/meshtastic/firmware/pull/5196)

If you want to use the heltec_bat() at v3.2 Boards you have to change from LOW to HIGH
`digitalWrite(VBAT_CTRL, HIGH);`

christianhubmann, 2025-01-30:
Can confirm. By the way, the schematic for v3.2 on Heltec's site links to the v3.1 version, but if you go to the [Downloadable Resources](https://resource.heltec.cn/download/WiFi_LoRa_32_V3/), there is a v3.2 schematic that shows the changed circuit for ADC Ctrl.

Stoli123, 2025-08-02:
I just acquired a couple of v3.2 boards and am unable to get heltec_vbat() to return anything other than .20 - .24, even with VBAT_CTRL set HIGH.