ropg/heltec_esp32_lora_v3 #71: Not so much an issue but

terraslate, 2025-01-01:
I'm not sure why the battery values read only to 254 on a scale of 0 to 4095? By my calculations you'd get 0.87v against a reference of 1.1v with a max voltage of 4.26 (I think that's the value in the header). Surely this value is 0.87/1.1x4095 is a lot larger than 254? And yet I also read values in esp-idf on the ADC gpio1 upto around 250 or so. Any ideas why?

ropg, 2025-01-01:
I'm not 100% sure what you mean, but I think the battery values of which you speak are deliberately squished into one byte simply to save table space that would be used for precision that would be overkill for predicting how long the battery will last. I [simply divide](https://github.com/ropg/heltec_esp32_lora_v3/blob/25791aa932a4ef4fe587834144ad6c5470c990ff/src/heltec_unofficial.h#L286) the highest and the lowest voltage measured by 256 to get a range of one byte.

terraslate, 2025-01-01:
So I'm porting your lib to esp-idf and using adc.h. I read analogue values directly from the pin and I noted that when I read them (after setting 37 low) I also get low numbers in a similar range to yours directly from gpio1. However the range should be 0 to 4095 where I expect 1.1v is 4095 with the attenuation of 0db. With the resistors R1 and R2 you'd expect the max would be 0.87v and therefore a full battery to be 3200ish when read. Maybe my maths is wrong or I don't understand this enough :-)