ropg/heltec_esp32_lora_v3 #75: Compatibility for HELTEC lora 32 v3.2

d4rkpunk69, 2025-01-29:
Can you provide a fix for new version of heltec lora 32 v3 version 3.2 because after uploading a sample code, idont know why it doest display anything or it just in sleep mode automatically. I found an article that need to set ADC_ctrl to high. Do you know how or wnyone?

djcasper1975, 2025-01-29:
I wonder why heltec did this. I have no 3.2 yet but I suspect it was goto with screen power because I have to use unofficial heltec library's to even get my own projects working. I remember it was a power issue with screen pin conflicts. I feel like I gotta buy a 3.2 now to see if my own lora projects work.

d4rkpunk69, 2025-01-29:
I fixed it by modifying your library, someone found a fixed and they shared it online. Found it on heltec forum

printman-wi, 2025-02-01:
for Version 3.2 you need:

// GPIO36 switches VEXT
const int VEXT_control =  36;   
const int ADC_control =  37;   

void setup() {
  unsigned status;
  byte value = 0;
  pinMode(VEXT_control, OUTPUT);
  pinMode(ADC_control, OUTPUT);
  
  heltec_setup();
  digitalWrite(VEXT_control, LOW); // VEXT ON ffor OLED display
  digitalWrite(ADC_control, HIGH); // allow ADC
  // Display
  display.println("Display works");
....

closed 2025-02-03 by d4rkpunk69