ropg/heltec_esp32_lora_v3 #77: Display issues on 3.2

whymusticode, 2025-02-10:
I started using this library for the 3.1, but on 3.2 the same code will work, but nothing shows up on the display. All the display examples are broken for the 3.2, but the examples using "HT_SSD1306Wire.h" work on 3.2

Sv443, 2025-02-20:
Issue #63 had the solution for me, it just needed a `heltec_ve(true);` after the `heltec_setup();`

lubeda, 2025-03-15:
On V3.2 Hardware you need also to turn **ve** on to use the display.

The probleme is this part in the header-file:

```
void heltec_display_power(bool on) {
  #ifndef HELTEC_NO_DISPLAY_INSTANCE
    if (on) {
      #ifdef HELTEC_WIRELESS_STICK
        // They hooked the display to "external" power, and didn't tell anyone
        heltec_ve(true);
        delay(5);
      #endif
```

ChiefBacon, 2025-03-16:
I have this issue as well.

EmbeddedDevver, 2025-03-22:
> Issue [#63](https://github.com/ropg/heltec_esp32_lora_v3/issues/63) had the solution for me, it just needed a `heltec_ve(true);` after the `heltec_setup();`

This fixed it for me.

IgnisAlienus, 2025-03-23:
> Issue [#63](https://github.com/ropg/heltec_esp32_lora_v3/issues/63) had the solution for me, it just needed a `heltec_ve(true);` after the `heltec_setup();`

This also fixed it for me. Thanks!